MatteCarra/AccA #224: Stopping deamon and restart it caused fatal bootloop

roflsunriz, 2022-12-14:
Environment
===========
Platform:Android
OS Version:8.1(EMUI 8.1)
Testing Software Version:AccA 1.0.34
Advanced Charging Controller v2022.6.4
Device Models:Huawei P20 Pro HW-01K CLT-L0J

Description
===========
Operated ‘stop’ Advanced Charging Controller Daemon(accd) from AccA then started it manually. Rebooted, then boot stucked at bootloader unlocked warning screen.

Steps to reproduce
==================
1.Install AccA
2.Notifys ACC installation fail
3.Install latest ACC manually
4.Restart AccA and confirm accd is working
5.Press ‘stop’ to stop accd via AccA
6.Sometime later, start accd via AccA
7.Reboot phone
8.Phone doesn't boot

Expected result
===============
Phone boots normally

Actual result
=============
Can not boot to Android
You can't mount any partition at TWRP Custom Recovery
You can't wipe any partition at TWRP Custom Recovery
You can't install/backup/restore zip/img at TWRP Custom Recovery
You can't ADB sideload/don't see real files at file manager/fix linux context/change file system/resize file system etc at TWRP Custom Recovery
You can't boot eRecovery(EMUI Recovery)
You can't enter into fastboot mode(still you can get into rescue&fastboot mode via connecting usb cable with pc and volume down + power buttons)
You can use ADB at TWRP but you can't use pull/push command, shows mount error
You can write img file via rescue&fastboot mode but the partition will not come back, console says writing successful but you do not have access to any partition yet.
You can't use dload to write firmware forcefully via USB-OTG USB memory because you don't have eRecovery

Reproducibility Rate
====================
Unknown, just did once

Workarounds
===========
Unknown

Severity (Reference)
====================
Critical

Priority (Reference)
====================
High

Screenshots / Videos / Logs
===========================
There is no way to take pictures unless you have extra phone or camera
The log appeared at TWRP was mostly mount error, like /system is invalid argument (error for every partition except recovery)

Notes
=====
I got bootloop at woke up. What I did before shutdown was just only AccA related, and this is exactly only related to system modifying (I'm not refering to system partition) probably make phone brick, so I made this issue. I tried to download firmware for CLT-L0J(Charlotte-L0J) because I didn't back up my phone but they are need to pay and have wired pay system, like paying to world local reseller or pay it with crypto-coins, so I could't buy them at all. If I have right Full-OTA firmware data maybe I could unbrick my phone. But the phone does not me allow to write/mount any partition so still hopeless anyway. I asked the cell phone store to repair it because I can't handle this anymore.

Dimoon12, 2022-12-30:
wow thats scary, good luck, maybe partitions corrupted? really wierd

roflsunriz, 2022-12-30:
I think so too. The partition table broken. The only saivor is format them but I don't know how. Where is [pt.blob](https://forum.xda-developers.com/t/solved-corrupted-partition-adb-and-fastboot-works.3096193/) and [gpt.bin](https://forum.xda-developers.com/t/partition-table-broken.4276897/)? I asked shop to repair then they answered it's 74,000 JPY to fix. I bought Pixel 6a for 32,000 JPY  and waiting to back my Huawei phone to my house now.