MatteCarra/AccA #238: Update ? Or is the app adandoned?

rubysm, 2023-09-14:
Acca is user friendly compared to acc, u shouldn't abandon this app.

fillwithjoy1, 2023-10-07:
what is there to update in this app? nothing.

rubysm, 2023-10-07:
Do u use an app that isn't being updated for long time which talks to system core circuitry modules, even this is a wrapper it may arises fear when using in costliest phone and also crashes few times,and a good dev would take easiest parts to user and harder parts to professionals.

fillwithjoy1, 2023-10-08:
> Do u use an app that isn't being updated for long time which talks to system core circuitry modules, even this is a wrapper it may arises fear when using in costliest phone and also crashes few times,and a good dev would take easiest parts to user and harder parts to professionals, even though I m familiar with command line it (acc) has high learning curve.

what the fuck? you shouldn't be paranoid anyways. just take a look at this diagram below

AccA <-> ACC <-> Battery Controller

As you notice, AccA is just a wrapper for ACC. You **should** keep ACC updated, but AccA is updated almost never. If you look at F-Droid, there was another version that was 2 years from this latest release. Unless ACC has a dramatic change in how commands are ran, there is **absolutely** no reason that this app needs to be updated.

Oh, and on another note, manufacturers (for me with Google at least) configure hard limits within the battery controller, so you and I cannot just make our batteries explode. So even then, it's not a big deal and there's nothing for you to be paranoid about :)

rubysm, 2023-10-08:
What the fcuk with u too, if u don't know about it fully go and install acc settings apk which doesn't works on new acc as it being not updated too both acca and acc settings app are being a wrapper also it crashes, u need to mind ur tongue before talking in a public forum.

fillwithjoy1, 2023-10-08:
> What the fcuk with u too, if u don't know about it fully go and install acc settings apk which doesn't works on new acc as it being not updated too both acca and acc settings app are being a wrapper also it crashes, u need to mind ur tongue before talking in a public forum.

lmao, you really need to learn more English to understand what I was saying. Moving on, the newest ACC version works with this app, I have got it running and working. 
Try these steps if it's not working:
1. Uninstall AccA 
2. Remove the ACC module
3. Restart
4. Install ACC and AccA and report back. 
And also remember that the person working on this is taking a break, so you can keep barking but that won't change anything - they are taking a break. And for the last statement, maybe learn some more english as well, there's lots of glaring grammatical issues in your statement so I don't feel that you know what you are saying either.

marbens-arch, 2024-12-13:
> what the fuck? you shouldn't be paranoid anyways. just take a look at this diagram below
>
> AccA <-> ACC <-> Battery Controller
>
> As you notice, AccA is just a wrapper for ACC. You **should** keep ACC [...]

Even that diagram isn't accurate a more accurate one is:
AccA <-> ACC <-> Kernel <-> Battery Controller

fillwithjoy1, 2024-12-14:
> Do u use an app that isn't being updated for long time which talks to system core circuitry modules, even this is a wrapper it may arises fear when using in costliest phone and also crashes few [...]

Again, what is there to update for?

weiluntong, 2024-12-21:
Lel, just gonna hop on this year old issue and also say that ACC Settings is also just a wrapper and not associated with ACC. I'd like to point out though that the lack of development has caused ACC to not use either GUIs. If not because of concrete reasons, then simply because the stalled development has caused distrust.

https://xdaforums.com/t/advanced-charging-controller-acc.3668427/

though, it *does* seem as though the project is well and truly abandoned, as people are finding that they need more and more workarounds to interface with ACC using this tool; and all communications from @MatteCarra seemed to have ceased.

Of course, I hope that's not the case and I'm simply mistaken, but there's certainly signs.

1993Banik, 2025-05-16:
Dear Developer,

The world of mobile health truly rests in your capable hands. The Advanced Charging Controller (ACCA) has been a lifeline for so many of us striving to extend our battery life and protect our devices.

We kindly urge you to continue development and update this powerful app. Most users are not familiar with how to operate the ACC module via command line, and your app makes this technology accessible to everyone.

Your generosity in creating and maintaining ACCA not only saves money—it saves countless mobile batteries from early failure. The world is watching, and we believe in your vision. I deeply admire your work and sincerely hope you will continue to lead this initiative for the greater good.

With respect and anticipation,
RB

fillwithjoy1, 2025-05-18:
> Dear Developer,
>
> The world of mobile health truly rests in your capable hands. The Advanced Charging Controller (ACCA) has been a lifeline for so many of us striving to extend our battery life and [...]

Oh hell nah, somebody take this man's ChatGPT subscription away

CortexReaver, 2026-09-29:
Have anyone tried this version: https://github.com/seyedehsanhadi/AccA ?