MatteCarra/AccA #249: I HAVE TRY BUILD NEW VERSION https://github.com/CRGBS/AccA-release/releases/tag/release

CRGBS, 2024-07-14:
This version includes forks from many people.
@bubendorf https://github.com/bubendorf/AccA
@ViniciusRed https://github.com/ViniciusRed/AccA
@MatteCarra https://github.com/MatteCarra/AccA

lovelyelfpop, 2024-07-19:
Great job!

theAkito, 2024-07-21:
Will this eventually be released on F-Droid?

CRGBS, 2024-07-21:
> Will this eventually be released on F-Droid?

probably not
Because I just collect other people’s forks for build

yejinson, 2024-07-21:
Thank you for your work. The custom settings are not working, except capacity control, can you fix it?

Carontes, 2024-07-21:
On LineageOS 20 it crashes as soon as it starts, so it's unusable for me. Hoping to see a new working version

theAkito, 2024-07-22:
> > Will this eventually be released on F-Droid?
> 
> probably not
> Because I just collect other people’s forks for build

Can still be released on F-Droid, if you re-label it and get the consent of all contributors.

ViniciusRed, 2024-07-22:
> Will this eventually be released on F-Droid?

Hello, when I have everything working maybe I'll publish a release on GitHub or F-Droid.

> On LineageOS 20 it crashes as soon as it starts, so it's unusable for me. Hoping to see a new working version

Yes, the code is totally broken, if you quickly change the dashboard menu, you will be able to see the other menus, some of the options are working, the profile part is not working properly, the theme is not working properly either and the translation is working more or less. 

Here's what the app looks like after @MatteCarra's last change. This version I'm using is the debug version from the dev branch. 

[image]
[image]
[image]
[image]
[image]
[image]

theAkito, 2024-07-31:
So, will there be one version, which most agree on?
That version likely needs to include all changes, although in the end it should also work. 😅

CortexReaver, 2026-09-29:
Also there is this: https://github.com/seyedehsanhadi/AccA