MatteCarra/AccA #253: Slower charging even after uninstalling App

Trtrtr70, 2025-01-10:
I have a Xiaomi Mix 4 that can charge fully from 0-100 in about 20 minutes (120w). After installing the app and setting it up I noticed slower charging, but after trying to charge faster and therefore deactivating the app it was still charging slower. Took at least 40 minutes.  I uninstalled the app and it is still charging slowly. The module in Magisk was also deactivated.

How can I get my phone to charge at default speed again?

VR-25, 2025-01-11:
This app is in maintained and severely outdated.

Plug the charger, boot into fastboot, then back into the system to reset the power management chip.

Trtrtr70, 2025-01-15:
Thank you for the fast reply! I read it already and just wanted to update that my USB Flex Cable ripped of my motherboard (destroying contacts) and I am currently unable to charge it via wire. I will update asap.